kowalej/BlueMuse #1: Open stream receives no timestamps and closes after 10 seconds

jdpigeon, 2018-05-27:
One of the students in the neuroBRITE program was unable to get BlueMuse successfully working on his computer (new Lenovo gaming PC).

He was able to install BlueMuse and view devices, but upon connecting, the Open Stream window displayed no timestamps received (or only a few timestamps coming in ~3hz). Then the stream would close after about 10s

jdpigeon, 2018-05-27:
Rebooting the computer fixed this issue

jdpigeon, 2018-06-22:
Can we reopen this?

I'm running into this error again, this time on my personal computer.

Rebooting the computer, turning bluetooth on and off, rebooting the app, turning the Muses on and off, nothing seems to be able to fix it this time.

Clicking 'Start Stream' Leads to the second win32 window opening, displaying 1 or 2 updates to the timestamp value, and then closing after a few seconds. The main BlueMuse window then crashes without any message after a few more seconds

jdpigeon, 2018-06-22:
One interesting thing I've found is that this error will switch between multiple Muses. With two available Muses, the app will always be in a situation where one of them will be able to start a stream and the other will not send any data and crash the app when attempting to open a stream. Note, this isn't only when trying to open multiple streams at the same time. Trying to connect independently, one after the other, still raises the issue.

Rebooting the bluetooth on the OS seems to be able to flip which Muse is working and which Muse isn't. Also, I'm pretty sure that the first Muse that is attempted after a refresh will be the one that will work

gibraillash, 2019-03-22:
Any updates to this issue? I've been having the same error but haven't found a solution

eliam, 2019-04-08:
Hi, I've also got this issue. Muse 2 - (303E).
bluemuse startup seems fine.  Device detection ok.  Start stream, opens a new window with a row for the device.  Window then closes and main app then crashes.  More than willing and able to help debug this, if you give me a hint on where / how to get started?

eliam, 2019-04-08:
ok. ...\LocalState\Logs folder doesn't exist.
...\LocalCache\Local\Logs just has the following: LSLBridge-Log-20190408.log

2019-04-08 22:17:49.557 [Information] LSL Bridge started.
2019-04-08 22:18:29.036 [Information] LSL Bridge started.
2019-04-08 22:23:03.511 [Information] LSL Bridge started.
2019-04-08 22:42:07.618 [Information] LSL Bridge started.
2019-04-08 22:42:50.444 [Information] LSL Bridge started.

johnty, 2019-04-18:
another user on Muse 2, Windows 10 (pro) running into the exact issue.

find device OK, hit start and new window opens with a single row; then closes and main app crashes.

kowalej, 2019-09-10:
Closing see #8 .

closed 2019-09-10 by kowalej