kowalej/BlueMuse #2: Bluemuse close automatically when i run it.

LiemFrans, 2018-09-06:


kowalej, 2018-09-06:
What version?

LiemFrans, 2018-09-06:
BlueMuse 1.0.8.0 .

kowalej, 2018-09-06:
Try restarting PC.

LiemFrans, 2018-09-06:
I've restarted it many times, but same.

kowalej, 2018-09-06:
Hard for me to diagnose - maybe provide more system info. Also try these things:

1. You can try right clicking the app from Windows menu > More > App Settings > Repair. 
2. Reinstall the app - ensure old version(s) are removed.
3. See if there are log files:

C:\Users\{Username}\AppData\Local\Packages\07220b98-ffa5-4000-9f7c-e168a00899a6...\LocalState\Logs\BlueMuse-Log-{Timestamp}.log

LSLBridge:

C:\Users\{Username}\AppData\Local\Packages\07220b98-ffa5-4000-9f7c-e168a00899a6...\LocalCache\Local\Locgs\LSLBridge-Log-{Timestamp}.log

LiemFrans, 2018-09-06:
this is bluemuse log
[image]

kowalej, 2018-09-06:
You are trying to launch from command line eh?

LiemFrans, 2018-09-06:
i trying in command line and trying to launch manual from start menu, but problem same.

kowalej, 2018-09-06:
Looking at the error it seems there is something wrong with the device being found. Device ID is used after finding a Muse during search. Are you sure your Muse is a 2nd generation - Bluetooth LE version. Are you sure the Muse works with other device(s)/software?

kowalej, 2018-09-06:
Do you have another Bluetooth device sitting around that has "Muse" in the name by any chance?

LiemFrans, 2018-09-06:
[image]
this is my devices list connected on my pc.

kowalej, 2018-09-06:
So I think the error is triggering from BluetoothManager.cs line 111:

[image]

I am able to force this error by putting some bad id data here when creating the BLE device object. Assuming this is where the error occurs on your machine - it is very strange that the device watcher would give a bad ID, considering it is supposed to find only BLE devices which are then filtered to make sure their name contains "Muse".

Just to make sure we are narrowing things down. Can you please try removing the Muse from your device list shown in the screenshot above and trying again.

LiemFrans, 2018-09-06:
Thankyou verymuch,, your apps can launch again... :) 👍 
[image]

closed 2018-09-06 by LiemFrans

kowalej, 2018-09-06:
As a note - add some try catch withing DeviceWatcher_Added method.