kowalej/BlueMuse #6: Persistent disconnects with version 1.1.0

jdpigeon, 2018-10-03:
BlueMuse version on a Windows 10 Surface Pro Tablet seems to pretty unreliable, often disconnecting within one minute. Unfortunately, there's no information in logs explaining the crash

Downgrading to 1.0.7.0 seems to solve this issue, which makes me think the bug might have been introduced with new feature development.

kowalej, 2018-10-03:
Try going to 1.0.9.0 because the latest version also had the UWP target version increased - no change in minimum version. However, nothing has really been changed with how the Bluetooth connection code works for many versions.

supcmd, 2019-01-28:
I am getting the same problem on all versions I have tried: 1.1.0, 1.0.9 and 1.0.7 crashes within 1 minute after connecting to MU-03.  Connects, and streams, then stops unexpectedly.

supcmd, 2019-01-28:
Thanks @kowalej for looking into this.  

Logs:  (I tried 4 times on 1.1.0.0)
 
LSLBridge-Log-20190128.log:
2019-01-28 17:28:30.857 [Information] LSL Bridge started.
2019-01-28 17:29:09.311 [Information] LSL Bridge started.
2019-01-28 17:29:51.365 [Information] LSL Bridge started.
2019-01-28 17:30:44.280 [Information] LSL Bridge started.

BlueMuse-Log-20190128.log
2019-01-28 17:26:48.257 [Information] BlueMuse started.
2019-01-28 17:29:05.790 [Information] BlueMuse started.
2019-01-28 17:29:40.234 [Information] BlueMuse started.
2019-01-28 17:30:38.558 [Information] BlueMuse started.

After pressing start streaming, I get the LSL Bridge Screen.  
It says 
Stream Info   |   Last Timestamp  | Rate (Hz)
Name: Muse-OC64(00:..) | 0 | 0

[image]

kowalej, 2019-01-29:
That's very strange, it's probably not throwing an exception since it would likely show up in one of the log files. You said you are using the Muse 2 correct? I don't have a Muse 2 to try myself so there may be an issue with it - I don't know for sure. Maybe try the regular Muse with your setup if you have access to one, or try the Muse with another computer...

kowalej, 2019-09-10:
Closing see #8 .

closed 2019-09-10 by kowalej