kowalej/BlueMuse #7: Epoch timestamps

bdyetton, 2019-01-10:
Hi,
I noticed some discrepancies in the timestamps between the 'time_series" values, and the 'first_timestamp' and 'last_timestamp'.
For example:
Last weekend (5th) I recorded some data. I've loaded this into python with pylsl (**which I hope is not automatically adjusting timestamps without my knowledge**). 

For the muse stream ```muse```:
```
muse['footer']['info']['first_timestamp'][0] != muse['time_series'][0]
```
muse['footer']['info']['first_timestamp'][0] == 1546716762.941125
muse['time_series'][0] == 1546545044.7723508
i.e. the 'time_series' values lead the 'first_timestamp' by 2 days and a few seconds

kowalej, 2019-03-10:
@bdyetton did you ever determine what the problem was here? There are some different timestamp formats you can choose from the app settings menu.

bdyetton, 2019-03-11:
Yea it was that BlueMuse uses epoch timestamps by default for it's stream, and other programs use the lsl default where timestamps are referenced from the computer startup time. When you load the xdf which contains multiple streams, they can't be synced properly. 
I would recommend having the bluemuse timestamps default be the lsl default going forward, or something more prominent in the docs (my bad for not reading them thoroughly)  
More discussion here:
https://github.com/sccn/labstreaminglayer/issues/13

kowalej, 2019-03-11:
Closing this issue - I may update the docs to clarify the default being Unix timestamps.

closed 2019-03-11 by kowalej