kowalej/BlueMuse #8: Problem connecting the device

Dazzid, 2019-01-31:
I am trying to connect to a Muse (1) using BlueMuse, using Windows 10 Education 1803. It recognizes the Muse device in the Bluetooth settings. After unpairing the Bluetooth device, I run "start bluemuse:" and it never gets the device, I might be forgetting a requirement before running blumuse?

kowalej, 2019-01-31:
The command you are running will only serve to start the app if not already launched. To have it connect and stream from the first Muse it finds you should use 
start bluemuse://start?streamfirst=true

There is also a command to connect and stream to a specific device. It's all on the readme page.

Dazzid, 2019-02-01:
ok, It makes sense. The problem I'm trying to point is that BlueMuse doesn't find any Muse device. However, Windows 10 can detect the device when I run "Add Bluetooth or other devices".

kowalej, 2019-02-01:
Are you using the original "classic" Muse that has 2 usb ports (one at each ear) or the one released 2016? The classic Muse isn't supported since it doesn't use Bluetooth LE. That would explain why the PC could pair but Blue Muse doesn't see it.

Dazzid, 2019-02-01:
Aaah! there is another version between the classic (with two USB) and Muse 2. Incredible I have two Muse versions and I can't really use them. 
Thanks

kowalej, 2019-02-01:
In theory it should work with the latest "Muse 2" but I don't have one to test with to verify. I think someone else was having issues with Muse 2 though, so perhaps there's a problem.

Dazzid, 2019-02-01:
Yes, the Muse 2 is recognized but BluseMuse crashes after starting to stream. Using start bluemuse://start?streamfirst=true

Dazzid, 2019-02-21:
Now I have the correct version of Muse, we bought one at the lab. My next issue is that BlueMuse is now seeing the device "Muse-F308 (...) but it keeps the "Status: Offline Streaming: No" Even if I start with start bluemuse://start?streamfirst=true. What might be the problem? Thanks for the help

kowalej, 2019-03-10:
Hey Dazzid, did you ever get this working?

Dazzid, 2019-03-20:
not yet, now I have the Bluetooth BLED112 dongle, to try in a Mac. In the Windows 10 (Nuc computer) I haven't managed to connect the Muse.

Dazzid, 2019-03-20:
what would be the correct way to install again BlueMuse app? Just following the steps described it will overwrite the installation?

kowalej, 2019-03-20:
Yea just run the PowerShell script on new version.

kowalej, 2019-03-20:
Also I'd like to know if you can pair with the Muse in your Bluetooth device menu on Windows.

Dazzid, 2019-03-20:
yes, that works normally

kowalej, 2019-03-20:
Ok and make sure you remove the paired Muse from that menu ("unpair it") before running BlueMuse.

Dazzid, 2019-03-20:
yes, I'm following all the instructions. The problem in the Intell Nuc is still the same as before, BlueMuse is now seeing the device "Muse-F308 (...) but it never passes the "Status: Offline Streaming: No"

kowalej, 2019-03-20:
Try resetting the Muse by holding down the power button for a minute. Also can you post your log files. You can finds the log location in settings menu.

Dazzid, 2019-03-21:
Now I manage to connect the Muse but after clicking on Start Streaming, it opens a new window and then it crashes.  
The BlueMuse-Log-20190321 file only shows this:
2019-03-21 14:24:07.201 [Information] BlueMuse started.

tsloan1377, 2019-05-12:
I am having the same problem connecting to BlueMuse with Muse2 in Windows10. 
I cannot connect via bluetooth to the computer (fails to pair). However I see the Muse listed in the BlueMuse window. 
When I connect, it streams for about 5 seconds, and opens a LSL bridge window, which then closes after seconds. The BlueMuse window says that one device is streaming, but then the program crashes. (This all happens within 5s).

As woith Dazzid, my log file just shows:
2019-05-12 10:27:28.633 [Information] BlueMuse started.

kowalej, 2019-05-12:
@tsloan1377 @Dazzid - when you guys first get the stream going and it opens the "LSL Bridge" window - did you get a prompt to let LSL through your firewall? If not - it might help to manually add this - and perhaps this is why the app is getting closed at the bridge stage...

If you go to Control Panel\System and Security\Windows Defender Firewall\Allowed apps > Change Settings > Allow another app > Browse > Enter the path to LSLBridge (it's probably C:\Program Files\WindowsApps\07220b98-ffa5-4000-9f7c-e168a00899a6_1.1.0.0_x64__5ejts32jjed8r\LSLBridge) > Select LSLBridge.exe > Click Add > Check Off Private and Public > OK

tsloan1377, 2019-05-13:
[image]
I have tried changing the firewall settings as you suggested, but it doesn't solve the issue. I've attached a screenshot of the LSL bridge and BlueMuse windows, taken in the few seconds they're open. Hopefully something there can help.

kowalej, 2019-05-13:
When the LSL bridge closes does the main window say the stream is still running? Also can you send me the log from both BlueMuse and the Bridge...

tsloan1377, 2019-05-13:
Yes,  it says streaming for a few seconds after the LSL bridge window closes, however BlueMuse exits or crashes) about 2-3 seconds later. 

Here are the contents of the log files from multiple attempts today:

Blue Muse:
2019-05-13 11:32:06.477 [Information] BlueMuse started.
2019-05-13 11:42:59.481 [Information] BlueMuse started.
2019-05-13 11:48:32.359 [Information] BlueMuse started.
2019-05-13 12:14:25.661 [Information] BlueMuse started.

LSL Bridge:
2019-05-13 11:42:43.082 [Information] LSL Bridge started.
2019-05-13 11:43:15.777 [Information] LSL Bridge started.
2019-05-13 11:50:13.976 [Information] LSL Bridge started.

tsloan1377, 2019-05-13:
I should also mention that it is able to properly detect the state of the headset. For example, if the headset is connected with the Android phone Muse app, it is considered offline by BlueMuse, and vice versa. Also, the searching behaviour of the bluetooth signal and the light patterns on the Muse are what I would expect: moving lights while Status is Offline, then solid lights when the Status is Online. 
Also as expected, I lose the ability to recognize the muse when Bluetooth switched off, but what is strange is that when I can detect the Muse in the windows bluetooth devices, it is never able to connect. 
Maybe this is related to tmy issue in BlueMuse.

kowalej, 2019-05-13:
If it shows as paired in Windows Bluetooth settings it definitely won't connect in BlueMuse. As for this random shutdown issue it seems as though the logs are not catching this issue at all. I'm starting to wonder.. could it be your antivirus software blocking the app at the LSL phase?

tsloan1377, 2019-05-13:
It was never able to pair in Windows Bluetooth, so it doesn't seem to be a problem of  multiple devices connecting to the Muse, it still occurs when I disable bluetooth on the phone, to reduce that possibility. Actually, my phone bluetooth also was never able to detect the Muse as a nearby bluetooth device, but within the app itself I could connect and the headset works.
 I haven't found a way to un-link the app from the headset, per se (in case this is the issue), except for signing out of the app.

I have added exceptions for both LSL bridge and BlueMuse.exe to windows firewall and Bitdefender, but my error still persists.

tsloan1377, 2019-05-14:
> Now I have the correct version of Muse, we bought one at the lab.

@Dazzid did you ever get this working with the other version of Muse? I'm wondering if I made a mistake to get the Muse2 instead of an older version.

Dazzid, 2019-05-14:
Hi @tsloan1377 you are describing in details the same situation I had. 
I'm currently working in some other research and I didn't try to get data from the device anymore.

ranibourbara, 2019-05-14:
same here! it crashes after few seconds. anyone managed to make it work?

kowalej, 2019-05-14:
@tsloan1377 and @ranibourbara to clarify are you both using the new "Muse 2" that launched like a few months back? If so I do not have one so I can't debug issues with that headset. If either of you are in Toronto maybe we could test one day.

tsloan1377, 2019-05-15:
I am using the Muse2, but I'm trying to return it to get the 2016 muse instead. 
I see on the NeuroTechX Slack channel that that's the version recommended for developing with. 
I'm in Montreal, but a trip to Toronto would be doable.

kowalej, 2019-05-15:
I'm gonna try to get my hands on a Muse2 and see if I can personally get it running in the next week or so.

Dazzid, 2019-05-15:
@kowalej that would be great!

ranibourbara, 2019-05-21:
@kowalej Hey thanks. 
no I am using the 2016 version. not muse 2. it connects starts streaming and disconnects after few seconds.

rschmaelzle, 2019-06-04:
Hi all
I don't want to add to the confusion, rather add to clarify:
I have two muses, one 2016 version (device says mu-02 though) , and the recently released Muse2 (device calls that mu-03).
When I connect them using BlueMuse, I can get the 2016 version to stream - all wonderful
When I try with the newer version (Muse2/mu-03), it starts the LSL bridge, but never shows sampling rate or any data, and then crashes after about 5 seconds.
Of note, since I also own a Mac, using the recently updated muselsl code (stream -PPG), i am able to get the newer muse (muse2/mu-03) to run. But since BlueMuse is so convenient to setup, it would be great to be able to run this with the newer models (including the ppg).
Best, r

kowalej, 2019-06-04:
Thanks for the info @rschmaelzle - it does seem like people are consistently having an issue with the Muse 2 where the LSL bridge crashes when the stream starts. I'm not sure what the cause of the issue is right now, but I'm trying to track down a Muse 2 for myself to help debug it.

kowalej, 2019-06-05:
I have created a test build with some updates which may help connectivity problems. You can download it here: [https://github.com/kowalej/BlueMuse/raw/master/TestDist/BlueMuse_1.1.1.0_Test.zip](https://github.com/kowalej/BlueMuse/raw/master/TestDist/BlueMuse_1.1.1.0_Test.zip)

This issue is difficult for me to debug because I don't have a Muse 2 and can't recreate the connectivity issues some of you are having. But I thought it may be helpful with this version to update the target SDK to the latest Win 10 kit which may help solve the issue. I also figured how to make the app explicitely pair to the Muse. To enable this forced pair mode, in the settings menu you can check off: "Always Pair Device". Then hit the Refresh button and you should be prompted to pair the device when it is detected.

[image]

tsloan1377, 2019-06-05:
Thanks @rschmaelzle for the clarification, I ordered an older Muse (2016) after returning my Muse2, and will try my luck with that one.

tsloan1377, 2019-06-07:
Just to follow up, Blueuse is able to connect and streams fine with the 2016 Muse.

nrcmichael, 2019-06-13:
I also have a Muse 2 and have run into the same problem described here by others. LSLbridge closes and then shortly after Bluemuse closes.

I tried installing the 1.1.1.0_Test version. With "always pair device" checked when I refresh Bluemuse a Pair Device window pops up with the message "Muse-24C1 Connection Failed".

Thanks for working at this @kowalej . I appreciate it's challenging when you don't have access to the device.

Has anybody successfully connected to a Muse 2 on Windows using any application?

kowalej, 2019-06-13:
@nrcmichael it's pretty interesting that even the pairing is failing then with the Muse 2... Not just the app specific code crashing.

nrcmichael, 2019-06-13:
@kowalej the pairing seems to be handled differently for the Muse 2. On all the Android and Windows devices I have tried to connect with via the system Bluetooth settings I can see the Muse 2 but when I try to pair it it fails. This is the case even with the Android device that can communicate with the Muse 2 using the Muse App. "Can't pair with Muse-24C1. Incorrect PIN entered."

kowalej, 2019-06-13:
Strange. I don't think there should be a pin at all for pairing the Muse

kowalej, 2019-06-13:
Maybe try a reset of the Muse by holding the power button for like 30 seconds...

dannyqp, 2019-06-27:
I have a Muse 2 and I try to run BlueMuse, but the same problems here occurs. LSLbridge closes without any data and then shortly after Bluemuse closes.
I tried any possible solutions without success. 

Thanks

eliam, 2019-07-13:
Hi, cloning your git repo.  I've got a muse 2, same issues as above.  How can I help debug this issue?

kowalej, 2019-07-13:
You should build and deploy the app in Visual Studio with the debug build profile. The you can go to Debut > Other Debut Targets > Debug Installed App Package.

See https://docs.microsoft.com/en-us/visualstudio/debugger/debug-installed-app-package?view=vs-2019

eliam, 2019-07-13:
cool, I'll have a go.  Looks like the worlds moved on a bit since my vs2013!  Grabbing 2019 community to try.

eliam, 2019-07-14:
Looks like a no go.  I've installed vs2019 and tried to build the solution from the git repo.  No dice, missing dependencies.  Thought maybe the UWP API stuff might be it but adding that via nuget just adds duplicate deps to xaml and other things.  Feels like trying to untangle a knotted ball!  FYI, I'm not a c# dev so maybe this stuff is normal for you but I would have thought pull + build would work! (-:

Severity	Code	Description	Project	File	Line	Suppression State
Error	CS0731	The type forwarder for type 'Windows.ApplicationModel.AppService.AppServiceConnection' in assembly 'Windows' causes a cycle	LSLBridge	C:\liam\dev\BlueMuse\LSLBridge\LSL\LSLStreamManager.cs	17	Active
Error	CS1069	The type name 'AppServiceConnection' could not be found in the namespace 'Windows.ApplicationModel.AppService'. This type has been forwarded to assembly 'Windows, Version=255.255.255.255, Culture=neutral, PublicKeyToken=null, ContentType=WindowsRuntime' Consider adding a reference to that assembly.	LSLBridge	C:\liam\dev\BlueMuse\LSLBridge\LSL\LSLStreamManager.cs	17	Active

aimarmun, 2019-08-28:
Hi!
I resolve this issue.
The problem with Muse2 is that the name are "Muse-40F8".
I will not elaborate much on the answer and I will expose the code that I have modified, in this way it will explain itself.

I have changed this code from the Constants.cs file:
```
public static readonly List<string> DeviceNameFilter = new List<string>()
        {
            "Muse", "SMTX"
        };
```
For this other:
```
public static readonly List<string> DeviceNameFilter = new List<string>()
        {
            "Muse", "SMTX", "Muse-40F8"
        };

```

And this other code from the Muse.cs file:
`if (name.Contains(Constants.DeviceNameFilter[1])) //line 105`

For this other:
`if (name.Contains(Constants.DeviceNameFilter[1]) || name.Contains(Constants.DeviceNameFilter[2]))`

As you will see the connection works perfectly with my Muse2:
[image]

I hope I have been helpful with my humble knowledge.

JeromeF91, 2019-09-09:
Thanks for the tip! I tried to compile it myself but without any success.

Hopefully @kowalej can provide a new version with the fix for the name :) 

> Hi!
> I resolve this issue.
> The problem with Muse2 is that the name are "Muse-40F8".
> I will not elaborate much on the answer and I will expose the code that I have modified, in this way it will [...]

kowalej, 2019-09-10:
@JeromeF91 thanks for playing around with the code to get it working on your machine. It seems like the logic you have added for you device basically forces it to stream like the Muse (Smith i.e. SMXT) glasses does. The fundamental difference was that the Muse had the AUX channel where  you could feed data in via USB, but the glasses didn't. I wonder - is this the problem with the Muse 2? I still don't have a Muse 2 - but I'm wondering, does it not have the AUX channel like Muse? This would explain why your logic worked.

VictorSaenger, 2019-09-13:
Dear @kowalej,
Thank you so much for building this package. I am also using MUSE 2 and experiencing the same disconnection problem. For all of us who are not familiar with compiling, would it be possible for you to share a .ps1 file with the changes prompted by @aimarmun? I understand we have to specify MUSE serial number, right? If this is too much to ask, could you provide some tutorials / compilers that I could use to try to do it myself? Thank you very much again for all your support!

JeromeF91, 2019-09-18:
@kowalej I'm not the one who found the workaround :) @aimarmun did.

Not sure about the AUX channel on Muse 2 sorry but if you want to get some code run on it, feel free :)

azadeh-haji, 2019-09-20:
I have the same issue with Muse 2. It does not seem to have an AUX channel @kowalej. Following here for any updates on this thread!

alexismorin, 2019-10-01:
Following this also, I'll see if I can fix it on my end for now!

alexismorin, 2019-10-03:
Okay, fixed it to a reasonable extent; Adding  "Muse-40F8" (and Muse-50CE for the most recent iteration of the muse 2) allows Bluemuse to start streaming through LSL, but the whole thing eventually crashes after 5-20 minutes with an Toggle Connection (false) exception in the logs. @kowalej Would that happen because Bluemuse tries to fetch some info that's in the smith glasses but not in the new musese? Anyways, here's the [WeTransfter link, YMMW ](https://we.tl/t-pfzB5jGiRZ)

VictorSaenger, 2019-10-03:
> Edit: Okay, fixed it to a reasonable extent; Adding "Muse-40F8" (and Muse-50CE for the most recent iteration of the muse 2) allows Bluemuse to start streaming through LSL, but the whole thing eventually crashes after 5-20 minutes with an Toggle Connection (false) exception in the logs. @kowalej Would that happen because Bluemuse tries to fetch some info that's in the smith glasses but not in the new musese? Anyways, here's the [WeTransfter link, YMMW ](https://we.tl/t-pfzB5jGiRZ)

Thanks for taking the time, @alexismorin ! Will try to run it on my computer and get back to you.

kowalej, 2019-10-03:
If the Muse2 does indeed have only the 4 channels we would want to sniff the Bluetooth to discern it from the other devices. Right now the detection just checks if "Muse" is in the device name or "SMTX" is in the device name for the glasses. For Muse 2 we would either check the firmware version, year, or attempt to connect to the extra channel first to see if it's available. @alexismorin keep in mind that each Muse has a unique ID which is the "-40F8" on your device. Everyone's device will have a different code on that part of the name. That's why @VictorSaenger can't connect using your code.

Unfortunately I still don't have a Muse 2 and probably can't work on this at all until next week at least. Sorry.

alexismorin, 2019-10-03:
All good, thanks for taking the time to work on this incredible project. You're in Toronto, right? If you're up for it I can ship you mine eventually for you to test with.

andrew-j-m, 2019-10-19:
This also works:
if (name.Contains(Constants.DeviceNameFilter[1]) || name.Contains(Constants.DeviceNameFilter[0]))

[6 more replies]

closed 2021-01-13 by kowalej