kowalej/BlueMuse #10: BlueMuse1.1.0.0 Installation Failed

AdityaSher, 2019-07-03:
Error: 
Either you need a new certificate installed for this app package, or you need a new app package with trusted certificates. Your system administrator or the app developer can help. A certificate chain processed, but terminated in a root certificate which isn't trusted (0x800B0109)

I have already installed the Certificate in the root folder, What must I do?

kowalej, 2019-07-03:
First off check that you are installing a recent version, old versions may have expired certs. Ensure you are running the Install powershell script as adminstrator, the script will take care of all the steps for you. Also, see the manual install guide in the readme if you are still stuck.  

Otherwise you can try installing this version:

https://github.com/kowalej/BlueMuse/raw/master/TestDist/BlueMuse_1.1.1.0_Test.zip

AdityaSher, 2019-07-04:
I was not following the Steps Properly hence Installation was Failing (Use local instead of Machine while installing certificate) I apologize for the trouble.

closed 2019-07-04 by AdityaSher

rschmaelzle, 2019-09-24:
Hi, I am receiving an error message complaining about an expired certificate. I am using the latest version - is that a known issue or is there an override? I couldn't find it in the other issues. Best, r