kowalej/BlueMuse #11: Problem big variation in the sample rate

CyrielleAlbert, 2019-08-20:
When I start streaming, after few seconds, the sample rate starts oscillating between low values (like 64) and high values (like 992). I've tried using another Muse and another computer but it's the same result. 
Does someone know where it could be from ?

kowalej, 2019-08-20:
If it's the same issue with 2 Muses, and 2 different computers, perhaps it is the network that is bottle necking it?

kowalej, 2019-08-20:
I'd also check the resource monitor on Windows while BlueMuse runs, see what kind of CPU and memory it's eating up, and compare with the system.

kowalej, 2019-08-20:
Sorry that I can't be of more help, but I haven't heard this reported before, so these would be the places to start looking. I will say though that I have run BlueMuse on relatively low spec machines, and it has run with consistent sample rate.

CyrielleAlbert, 2019-08-20:
Thank you for your answer. I'll check that.

closed 2019-10-19 by kowalej