kowalej/BlueMuse #12: Errors in Visual Studio solution

nrcmichael, 2019-08-29:
I have downloaded the solution and attempted to open it in Visual Studio 2017 but am running into errors. I installed the NuGet Package Microsoft.Windows.SDK.Contracts as suggested by the page linked in the readme.

The main errors appear to be coming from LSLStreamManager.cs. with 'type or namespace could not be found' for the following:
AppServiceConnection
AppServiceRequestReceivedEventArgs
ValueSet

also, 'The name 'Package' does not exist in the current context.'

Possibly related to this, VS indicates that for the following commands the using directive is not necessary:
using Windows.ApplicationModel;
using Windows.ApplicationModel.AppService;
using Windows.Foundation.Collections;

kowalej, 2019-09-10:
Those should be in the Window.dll - unless you are using an old version of Windows. Can you tell me what your system is running?

nrcmichael, 2019-09-10:
I'm running Windows 10 Version 1803. I also tried it on ver. 1809 and get the same error.

andrew-j-m, 2019-10-19:
Managed to get it working by uninstalling NuGet package UwpDesktop and manually referencing:
C:\Program Files (x86)\Reference Assemblies\Microsoft\Framework\.NETCore\v4.5\System.Runtime.WindowsRuntime.dll
C:\Program Files (x86)\Windows Kits\10\UnionMetadata\Windows.winmd

kowalej, 2019-10-19:
I updated a bunch of the solution settings, see if you can build now by pulling latest changes.

andrew-j-m, 2019-10-20:
The solution probably is download Visual Studio 2019.
FYI - configuration is for Visual Studio 2017 Community edition:
Configuration Manager x64
Target Platform Version 10.0.15063.0 in three projects
Platform Toolset Visual Studio 2017 (v141)
Uninstall UwpDesktop package
Add reference to C:\Program Files (x86)\Reference Assemblies\Microsoft\Framework\.NETCore\v4.5\System.Runtime.WindowsRuntime.dll
Remove reference to Windows
Add reference to C:\Program Files (x86)\Windows Kits\10\UnionMetadata\Windows.winmd

nrcmichael, 2019-10-24:
Thanks andrew-j-m and kowalej. After installing VS2019 and following Andrew's instructions I got very close but there's still one error left I can't seem to get rid of:

"No certificate found with the supplied thumbprint: ..."

I tried deleting this line in BlueMuse.wapproj:
<PackageCertificateThumbprint>5635D664D342A777F406A6827D17F05AFACAA808</PackageCertificateThumbprint>
but that didn't help.

nrcmichael, 2019-10-24:
OK, I figured out how to generate my own certificate and got it to build.

Thanks for the help.

closed 2019-10-24 by kowalej