kowalej/BlueMuse #13: Build errors

zeos, 2019-09-01:
Hi,

thanks for developing this! 
I am trying to build the Bluemuse, but I get this:

```
Severity	Code	Description	Project	File	Line	Suppression State
Error	CS0731	The type forwarder for type 'Windows.ApplicationModel.AppService.AppServiceConnection' in assembly 'Windows' causes a cycle	LSLBridge	C:\Users\zeos\Source\Repos\BlueMuse\LSLBridge\LSL\LSLStreamManager.cs	17	Active
Error	CS1069	The type name 'AppServiceConnection' could not be found in the namespace 'Windows.ApplicationModel.AppService'. This type has been forwarded to assembly 'Windows, Version=255.255.255.255, Culture=neutral, PublicKeyToken=null, ContentType=WindowsRuntime' Consider adding a reference to that assembly.	LSLBridge	C:\Users\zeos\Source\Repos\BlueMuse\LSLBridge\LSL\LSLStreamManager.cs	17	Active
Error	CS0731	The type forwarder for type 'Windows.ApplicationModel.AppService.AppServiceConnection' in assembly 'Windows' causes a cycle	LSLBridge	C:\Users\zeos\Source\Repos\BlueMuse\LSLBridge\LSL\LSLStreamManager.cs	59	Active
Error	CS1069	The type name 'AppServiceConnection' could not be found in the namespace 'Windows.ApplicationModel.AppService'. This type has been forwarded to assembly 'Windows, Version=255.255.255.255, Culture=neutral, PublicKeyToken=null, ContentType=WindowsRuntime' Consider adding a reference to that assembly.	LSLBridge	C:\Users\zeos\Source\Repos\BlueMuse\LSLBridge\LSL\LSLStreamManager.cs	59	Active
Error	CS0731	The type forwarder for type 'Windows.ApplicationModel.AppService.AppServiceRequestReceivedEventArgs' in assembly 'Windows' causes a cycle	LSLBridge	C:\Users\zeos\Source\Repos\BlueMuse\LSLBridge\LSL\LSLStreamManager.cs	59	Active
Error	CS1069	The type name 'AppServiceRequestReceivedEventArgs' could not be found in the namespace 'Windows.ApplicationModel.AppService'. This type has been forwarded to assembly 'Windows, Version=255.255.255.255, Culture=neutral, PublicKeyToken=null, ContentType=WindowsRuntime' Consider adding a reference to that assembly.	LSLBridge	C:\Users\zeos\Source\Repos\BlueMuse\LSLBridge\LSL\LSLStreamManager.cs	59	Active

```

any help is appreciated

best
--zeos

kowalej, 2019-09-10:
Sounds related to #12 - maybe you guys should try removing all the Windows.xxx references and re-adding and see what has changed. Please let me also know what Windows build you are using.

kowalej, 2019-10-19:
I updated a bunch of the solution config, you can probably build it now if you pull the latest changes.

kowalej, 2019-10-19:
Closing, refer to #12 .

closed 2019-10-19 by kowalej