kowalej/BlueMuse #14: Tried to install, the .cer has expired

jscodedeep, 2019-10-01:


kowalej, 2019-10-03:
@jscodedeep what version - cert should expire next year with the latest one. Unfortunately for dev certs I can only able to make them 1 year expiry.

jscodedeep, 2019-10-03:
Thanks, found 1.1.1 version under TestDist. Was only looking under the Dist.

oori, 2019-10-04:
Installing v1.1.1.0 from `TestDist`, I'm getting this error while installing manually.  installing certificate did work, this error comes later, while installing the app:

> Either you need a new certificate installed for this app package, or you need a new app package with trusted certificates. Your system administrator or the app developer can help. A certificate chain processed, but terminated in a root certificate which isn't trusted (0x800B0109)

[image]

kowalej, 2019-10-19:
New version coming today will have 2 year valid cert expiring 2021.

closed 2019-10-19 by kowalej