kowalej/BlueMuse #16: Muse (1) is visble but can't connect

nukadelic, 2019-11-02:
I don't have the classic ( original ) muse with 2 usb ports, mine is a single usb port with a power button on the right side ( together with the LED lighs and the USB connections )

It will detect the device but will fail to connect 

[image]

nukadelic, 2019-11-02:
Just tried setting to always pair and it did try and connect but failed 

[image]

kowalej, 2019-11-02:
Start by trying to pair the device with your PC normally. Go to Start > Search "Bluetooth and Other Devices Settings" > Click "Add Bluetooth or other devices" > Bluetooth.

You may also try pairing your device with the official Muse app on a smartphone first - just to rule out connection issues with the unit itself.

Also try resetting the headset by holding down the power button for about 30 seconds.

nukadelic, 2019-11-03:
It pairs with the official app on the smartphone, seems like its a windows issue, i won't find it before or after resetting the headset by holding the power button

Zyx-Z-Y-X, 2019-11-21:
I got the same problem 
my MU-02 can connect to my phone apps , can connect and be detected by BlueMuse x64 on PC_2
BUT
can't be detected or visible on PC_1
BlueMuse x86
The log file say : 
  [Information] BlueMuse started.
  [date] [Error] Exception during find device (DeviceWatcher_Added) (device ID=BluetoothLE#BluetoothLE20: ...)

nukadelic, 2019-11-24:
Dropped my headset, now its broken. I'm out, peace.

closed 2020-01-02 by kowalej