kowalej/BlueMuse #17: Streaming other sensor data from Muse 2

CChilli, 2019-11-18:
From what I understand, Muse 2 also provides gyro, ppg, and accelerometer data. Is streaming those data possible using BlueMuse? I am using muse-lsl for recording the data.

rschmaelzle, 2020-01-08:
hi all, i just ran into this as well. I am using bluemuse (1.1.10) with the new Muse2. The eeg-channels are transmitted to LSL and ade recorder, but PPG doesn't show up in the XDF-file. I uinderstand that BlueMuse doesn't support this yet, but wanted to ask whether there are plans to integrate. Over at the MuseLSL package it seemed that it is [possible to hand the PPG-stream](https://github.com/alexandrebarachant/muse-lsl/pull/93) over to the labstreaminglayer, but I don't understand how to do that. I am using BlueMuse for the sake of easyness - if it is possible to do this with the muselsl package and a BLED112 dongle, that could be done. Also, I learned that Muse is putting out a new headband, which seems pretty interesting, and the  question would be whether there are any plans to integrate this (which also contains EEG, PPG, gyro etc..). Thank you very much, Ralf

kowalej, 2020-02-12:
Working on adding PPG, accel, and gyro.

closed 2020-02-12 by kowalej