kowalej/BlueMuse #18: Muse not visible but show presence when always pair

Zyx-Z-Y-X, 2019-11-21:
Hello There,
I got a new muse model : MU-02 (look like 2016 version).
BlueMuse working well with one PC but do not on the other, missing to show the muse in the detected listed device.
Below my initial Tchek-Up :
  
Phone : Bluetooth Setting System :
<table>
 <tr><td>Detected : </td> <td>1</td></tr>
</table>
Phone : Muse Apps :
<table>
 <tr><td>Detected : </td> <td>1</td></tr>
 <tr><td>Pair : </td> <td>1</td></tr>
 <tr><td>Running : </td> <td>1</td></tr>
</table>
PCx64 : Bluetooth Setting System :
<table>
 <tr><td>Detected : </td> <td>1</td></tr>
</table>
PCx64 : BlueMuse and Scrips :
<table>
 <tr><td>Detected : </td> <td>1</td></tr>
 <tr><td>Pair : </td> <td>1</td></tr>
 <tr><td>Running : </td> <td>1</td></tr>
</table>
PCx86 : Bluetooth Setting System :
<table>
 <tr><td>Detected : </td> <td>1</td></tr>
</table>
PCx86 : BlueMuse  :
<table>
 <tr><td>Detected : </td> <td>0</td></tr>
 <tr><td>Pair : </td> <td>0</td></tr>
 <tr><td>Running : </td> <td>0</td></tr>
 <tr><td>Always Pair "Experi-Mental" : </td> <td>1</td><td>Show ID but Fail To Connect</td></tr>
</table>

**The Log file at :** 
SystemDrive\Users\USERNAME\AppData\Local\Packages\07220b98-ffa5-4000-9f7c-e168a00899a6_te84kyc8jyx80\LocalState\Logs\BlueMuse-Log-[Date].log

Show Me Something Like This :
[Date] [Information] BlueMuse started.
[Date] [Error] Exception during find device (DeviceWatcher_Added) (device ID=BluetoothLE#BluetoothLE20:[Address]).

**and Visual Studio at:**
Debug > Debug Installed Apps Package > BlueMuse : 
Finally give me this : 

[image]

Got this wonderful exception 
0xC0000005: Access violation reading location 0x00000000.

and I admit , I'm a bit stuck at this point ...

_Note : kowalej You are my actual Top-IRL-Hero-for-"The-Years-2019"_

Izacht13, 2022-03-10:
Same issue here

kowalej, 2022-03-11:
Not sure what I can do to help debug this. Probably an issue with Bluetooth stack hardware access. Some further information on hardware would be appreciated.

closed 2026-08-15 by kowalej