kowalej/BlueMuse #19: Exception during toggle stream (start=False).

alexismorin, 2019-12-31:
Hey! It's me again.

So after all the shenanigans in [this issue](https://github.com/kowalej/BlueMuse/issues/8), I managed to get my Muse 2 to connect and work... briefly.

After an indeterminate ammount of time after I start streaming (with LSLBridge apparently working properly), LSLBridge shuts down, with no error message in the logs and only _[Error] Exception during toggle stream (start=False)._  in the BlueMuse log. 

Anything I missed on my end? I made sure I dragged both required DLLs in my system32, that all dependencies were installed properly and that my Bluetooth wasn't shutting off due to power consumption stuff.

Happy New Year!

kowalej, 2020-01-02:
Are you able to record any data? Are you using the Muse at the same position while recording, or are you moving away from the PC? If it's a random disconnect it could be signal related.

synicalsyntax, 2020-01-09:
BlueMuse was previously working fine for me until I suddenly got this error as well. This is what shows up in the logs:
```
2020-01-09 15:27:25.984 [Error] Unexpected null GATT characteristic (UUID=273e0003-4c4d-454d-96be-f03bac821358) during toggle stream (start=True).
```

synicalsyntax, 2020-01-09:
Update: I reset the Muse headband by holding the power button for about 30 seconds (the lights should blink to indicate it was successfully reset) and now the error is gone. Hope that helps!

closed 2020-02-04 by kowalej