kowalej/BlueMuse #20: PPG for BlueMuse

rschmaelzle, 2020-01-15:
Hi all, 
i just wanted to inquire whether there are any updates/recent plans about integrating the PPG-datastream from the new muse2 system into the BlueMuse. I've referenced the issue below, which was brought up earlier. Thanks a lot and 
Best, Ralf

hi all, i just ran into this as well. I am using bluemuse (1.1.10) with the new Muse2. The eeg-channels are transmitted to LSL and ade recorder, but PPG doesn't show up in the XDF-file. I uinderstand that BlueMuse doesn't support this yet, but wanted to ask whether there are plans to integrate. Over at the MuseLSL package it seemed that it is [possible to hand the PPG-stream](https://github.com/alexandrebarachant/muse-lsl/pull/93) over to the labstreaminglayer, but I don't understand how to do that. I am using BlueMuse for the sake of easyness - if it is possible to do this with the muselsl package and a BLED112 dongle, that could be done. Also, I learned that Muse is putting out a new headband, which seems pretty interesting, and the  question would be whether there are any plans to integrate this (which also contains EEG, PPG, gyro etc..). Thank you very much, Ralf

_Originally posted by @rschmaelzle in https://github.com/kowalej/BlueMuse/issues/17#issuecomment-572157271_

kowalej, 2020-01-22:
I'd like to add PPG, but still don't have a Muse 2 or Muse S to work with.

rschmaelzle, 2020-01-22:
hi @kowalej - thank you.  the way I see it there are at least two options: 
i) either we get you one 
 [;-) i'd be totally willing to chip in for that! Or: perhaps we can get the company to give you one given that your software is clearly so much better than the cumbersome tool they offer ;-)   Or: there's also a money-back guarantee ]
ii) can you specify what would need to be done? 
I am no big developer, but I may get this done...or at least help/advance. As said, i've been able to connect the muse2 to the LSL layer and transmit the EEG-streams, i just don't know what it would take to "serve" the PPG stream (i.e. to plug it into the LSL and transmit those data) - currently the Muse2 does connect to lsl and I can record XDF-files, but those only contain four eeg streams.
best, ralf.

kowalej, 2020-02-04:
@rschmaelzle I have a Muse 2 on loan now, so I'll be working on adding the PPG in the next week or so.

kowalej, 2020-02-14:
@rschmaelzle I have a test build up now with PPG, Accelerometer, Gyro, Telemetry available and some other new features. I still need to do a bit of testing, but if you want to give it a try and provide some feedback you can download it here: https://github.com/kowalej/BlueMuse/raw/enable-ppg-accel-gyro/TestDist/BlueMuse_2.0.0.0_Test.zip

kowalej, 2020-02-19:
Closing - download 2.0.0.0 https://github.com/kowalej/BlueMuse/raw/master/Dist/BlueMuse_2.0.0.0.zip for PPG support.

closed 2020-02-19 by kowalej

rschmaelzle, 2020-02-19:
@kowalej THANK YOU SO MUCH, jason! I will check it out tomorrow! Thanks a lot for all the effort! If you ever make a patreon account or so, let me know!