kowalej/BlueMuse #21: Problem connecting Muse 2016 Device

HVjay, 2020-01-27:
When listing on BlueMuse, the device is shown as Status:Offline Streaming: No. I am using Windows 10 and it does not detect the device on bluetooth. I attached the log files as well, please take a look. 
[BlueMuse-Log-20200126.log](https://github.com/kowalej/BlueMuse/files/4115426/BlueMuse-Log-20200126.log)

kowalej, 2020-02-04:
Can you try manually starting the stream by clicking the button in the interface? Not sure about these logs, they don't indicate much.

HVjay, 2020-02-23:
The button is greyed out and I am unable to click it. I recently tried it again with BlueMuse 2.0 and am receiving the same error.

kowalej, 2020-02-23:
If the button is greyed out, you should see Status:Offline, and you cannot stream, since BlueMuse was not able to  establish a connection to your device. Some things to try:

1. Press "Windows" button > search "Bluetooth & other devices" > Scroll down to Other Devices, and ensure your Muse is not in that list. If it is please remove/unpair it and try BlueMuse again.

2. Hold the power button on your Muse for about 20 seconds and see if it connect after letting go.

3. In BlueMuse settings enabled Always Pair Device. Then hit refresh in BlueMuse, if it picks up your device it should ask to pair with it.

Another thing I'm curious about is if this Muse has been able to connect to other applications, like the official Muse app on iOS or Android.

OfirKedem, 2020-03-14:
hi,
I have a similar issue. I have tried your suggestions but they didn't work. 
I **can** connect to other applications.
It's important to note that sometimes it manege to pair, but even then when I press the stream button the streams stops after a few seconds.

kowalej, 2020-03-15:
Are you running the latest version (2.0.0.0)? You can find it here if not: https://github.com/kowalej/BlueMuse/releases

OfirKedem, 2020-03-15:
Yap. Downloaded it yesterday. I'm using Windows 10.

jbfaulhaber, 2020-04-05:
Same problem here. Bluemuse recognizes my Muse-2 but can not complete the pairing. 
For grins, I tried to pair with Win10. Again, it was recognized but the pairing failed.  
The log clearly shows my Muse-2 id. Exception during find device (DeviceWatcher_Added) (device ID=BluetoothLE#BluetoothLE3c:77:e6:ef:d2:e9-00:55:da:b7:4d:8d).
System.NullReferenceException: Arg_NullReferenceException

I am calling Bluemuse from startMuseStream.

The headband connects to MindMonitor on my smartphone with no problem.

kowalej, 2020-04-05:
Guys, I wish I could help, but it seems like some Win10 machines just have janky Bluetooth - the log unfortunately doesn't point to anything useful - it's very odd that it can pickup the device, but then renders a NullRef. 

One theory is that another device is competing for that connection (such as your smartphone) at the same time. Other than the steps I listed above, I would perhaps try turning off the phone's Bluetooth if it is nearby and try your PC again.

jbfaulhaber, 2020-04-06:
THX - I'll try it tomorrow
I also have a case where I have powered off the cell phone, reset the Muse 2 and It asks for the pin. This also happens in just plain Android. I have tried 0000 and 1234.
I've downloaded the 2.0.0.0 Bluemuse and will load it tomorrow. Does it have the new constants? I think it is from 2/24/20/

Great minds discuss ideas; Average minds discuss events; Small minds discuss people
Eleanor Roosevelt

kowalej, 2020-04-06:
2.0 adds support for accelerometer, gyro, PPG, and has a bunch of little tweaks. In regards to it asking for a PIN - that is very bizarre, and not supposed to happen at all... one thing you can try with BlueMuse 2.0 is (if the device shows up at all) you can click "reset" which will send a special command to the Muse which may resolve this PIN issue.

chrislos, 2020-04-28:
Hey everyone. Maybe it's a trivial solution. But it worked for me. I just installed BlueMuse on a fresh setup Winodws10 machine inlcuding Intel BT5.0 on an ASUS Mainboard. Couldn't pair an Original Muse with the system properly. (Model Name: unknown, start streaming greyed out etc. ) Even Bluetooth Drivers didn't show any conflicts I just updated them. Now it's working properly.

amisepa, 2022-01-17:
Hi, any update on this issue. I have the same problem with Muse 2016, latest Bluemuse version. I can only get the "stream" button operational if I check the "always pair device" box. But then it crashes as soon as I click try to stream. The device does pair on other Apps (mindmonitor, musedirect).

Is it the Windows Bluetooth that is too weak? Would a dongle help?

Thanks

q8wwe, 2026-02-11:
2026-02-11 01:43:49.769 [Error] Exception during find device (DeviceWatcher_Added) (device ID=BluetoothLE#BluetoothLE98:3b:8f:b9:a8:e7-00:55:da:b9:8a:14).
System.NullReferenceException: Object reference not set to an instance of an object.
   at BlueMuse.Bluetooth.BluetoothManager.<DeviceWatcher_Added>d__21.MoveNext() + 0x545

Im having this error any fix

closed 2026-08-15 by kowalej