kowalej/BlueMuse #24: Installation Error in parsing the app package

emanuelgo, 2020-03-22:
Hi Jason, 
thank you very much for creating BlueMuse and also finding a way for integrating the PPG sensors! I just tried to install your Bluemuse 2.0.0.0 install bundle but the automatic power shell installation, as well as the manual installation, didn´t work. On each appx file, I run into the error shown in the attached screenshot: Error in parsing the app package 

Any idea how I could solve it?
WIN10 64bit

[image]

And did you already try to connect Bluemuse with the Muse S as well?

Thx and all the best,
Emanuel

kowalej, 2020-03-22:
You may have to open an administrator Powershell, cd to the InstallBlueMuse script and run. I'm not sure why you would get this parsing error - I have installed this on several Win10 64-bit machines. Maybe it is even possible that the download or extracted archive got corrupted? I'd try running as admin first though.

kowalej, 2020-03-22:
Also - this has not be tested with Muse S - I don't have a Muse S, but from what I have gathered the Bluetooth configuration may be a little bit different on it, so I'm not sure it would be compatible without software changes.

emanuelgollob, 2020-03-25:
I found an error in my registry :/ after repairing it everything works now. Thanks for your help!

closed 2020-04-05 by kowalej