kowalej/BlueMuse #25: About PPG signal

iPsych, 2020-03-27:
Dear @kowalej,

Thanks for awesome app which interaxon never provides.
I am interested how do you process PPG data.
Using muselsl, There are three streams (PPG1, PPG2, PPG3). 
Do you developed or figured out the way to generate PPG wave from those three datastream?

kowalej, 2020-04-05:
At present, I added PPG on request from other people, but have yet to do PPG related research myself. I would recommend asking around the NeurotechX Slack channel (if you aren't a member you can sign up here: https://docs.google.com/forms/d/e/1FAIpQLSfZyzhVdOLU8_oQ4NylHL8EFoKLIVmryGXA4u7HDsZpkTryvg/viewform). The PPG signal is managed the same way as muse-lsl (Python library) handles it, so it may be worth asking people who are using that software as well.

closed 2020-04-05 by kowalej

ErikBjare, 2020-10-26:
@iPsych Did you ever figure this out? Curious about this too.