kowalej/BlueMuse #26: crash on refresh, when device is switched -off, while running

zeos, 2020-04-14:
Hi,

Thank you for developing the BlueMuse!!! I noticed a bug, resulting in a crash, when one switches off the Muse Device, and presses refresh.

kind regards

kowalej, 2020-04-14:
Hey thanks for the info. I have encountered this bug from time to time, it seems to happen randomly, so I'm not sure how to approach a fix. If you don't mind - can you post your logs from BlueMuse - they may provide some insight. The location of the log files is available in the settings menu of BlueMuse at the very bottom (you may have to scroll down). There should be a box with the directory where the logs are.

zeos, 2020-04-14:
[BlueMuse-Log-20200414.log](https://github.com/kowalej/BlueMuse/files/4477566/BlueMuse-Log-20200414.log)
Here is my logfile. Thank you for taking a look at it!
In my case 100% consistent:

1. Start BlueMuse
2. Connect to an active Muse device
3. Start streaming
4. SwitchOff the device
5. After ~4=5s BlueMuse crashes 

or press [refresh] right after the device is switchedOff -> crash

kowalej, 2020-04-23:
Thanks for the log, I should be able to handle this error in the next build. I'll leave this open until then.

antonello87, 2020-12-21:
Hi @kowalej, 
do you have any updates on this? Do you plan to release a new version in short time or have an indicative date?

I checked @zeos log and I also, sometimes, get the "Exception during toggle stream (start=False). Exception message: ObjectDisposed_Generic." error, especially when I try to stream from multiple muses.

P.s. Thanks a lot for this amazing app!!. It works really well and I think we are very close to have a really stable version.

closed 2026-08-15 by kowalej