kowalej/BlueMuse #27: OSC streaming

zeos, 2020-04-14:
Is it possible to integrate an OSC streaming capability, as it was supported by the (now discontinued) MuseDirect. There is also a vast infrastructure and software actively using the OSC (OpenSoundControl) protocol.

There is a full specification about the OSC messages. 
Some of them, I suppose, use statistics(StdDev) to stream the quality of the signal, FFT (PSD) separated in bands, etc....  

best regards

kowalej, 2020-04-14:
OSC would be possible, I can assure you of that - however, I don't know much about that protocol. Is there a defacto C# library or wrapper to use for interfacing with the OSC? What does the data look like when streamed from Muse Direct? Is it split by channel? How are the streams named?

I'm definitely open to pull requests for such a thing, I'm not sure I'd want to take the time to build it myself - honestly you are the first person to explicitly ask for OSC, but that's not to say the demand isn't there in the community. Although, from what I have seen a lot of people stick to LSL for this type of data.

zeos, 2020-04-14:
Hi! Thank you for your answer and interest in my suggestion. 

There is an entire community and many projects around the Muse OSC streaming (mainly form the Windows app MuseDirect and the musio-io (OSX) that are entirely based on the OSC communication. 

The OSC protocol is widely used in the digital arts field and is very popular protocol for communication between apps. It is simple and robust. Here is the specification: http://opensoundcontrol.org/introduction-osc

Here is the OSC messages for the Muse: https://mind-monitor.com/FAQ.php#oscspec

This is a .NET OSC library: https://github.com/dastevens/CoreOSC

The OSC bundles have timestamps and can hold a group of messages

best,

oori, 2020-04-22:
I'm half way through this path.        The main values are easy to emulate (absolute eeg power, accel, average relative power, etc..), but with the things like "Touching Forehead" and "horseshoe", it's harder to emulate the same results muse SDK gave.  for example see: https://github.com/kowalej/BlueMuse/issues/23#issuecomment-591857093

I'm not convinced if it should be integrated in BlueMuse or a LSL2OSC app,   but I agree with @zeos - there are many "old" apps waiting for a muse-io/musedirect replacement.

kowalej, 2020-04-23:
I almost think a generic LSL -> OSC intermediary library could be useful. Of course it would introduce additional latency, but I doubt this matters much for most people who are trying to record data and process offline VS real time stuff. It would also work for people who stream from muse-lsl or otherwise. @oori are you looking to spearhead a project like that?

oori, 2020-04-24:
My "muse direct emulator" works with `muse-lsl`.  written in python, it's a WIP, far from publishable, and has some über-specific customisations for our needs.   I would share it on gist as reference, not start a library with that.

Two things:

1. The _thing_ isn't protocol conversion `lsl2osc`, but the eeg processing pipeline.   those numbers and algorithms are "opinionated"  (e.g. averaging methods, factors, window sizes, ..). that highly affects the end user experience.  The output curves are different.              It can be resolved with presets, or pluggable pipeline (plugins).

2. Another app or a button in Bluemuse?      
we hackers, build our own tools, and can easily run 4 scripts in parallel.  But, I think the idea echoed by @zeos, is that there's a bunch of abandoned muse users in "maker/art/other scenes", who don't write code and only OSC protocol exists in their devices/software.  They'd love to get a muse-io / musedirect-windows replacement.       Your "Bluemuse" get's a magic checkbox `send osc`,  or it's some multi-platform CLI.. 

I'm currently busy with another development,  but could somehow be assistive in this effort.

guibot, 2020-07-15:
I would totally vote for OSC capabilities with BlueMuse.

Two years ago I (and my team) have developed an interactive installation using four Muse headsets with the original Muse Direct streaming OSC to MAXMSP.

Now we want to revisit the project and Muse Direct has been removed from the developers site, and we came across BlueMuse. 
Unfortunately, I can't do anything with LSL, I cannot even connect it to MAXMSP, and I can't find any LSL to OSC converter.

If any of you could add OSC to the package I believe it will be highly appreciated by the community.

kowalej, 2020-07-16:
I think it would be tricky to add the OSC support directly within BlueMuse considering the way things are currently architected.

I would recommend taking a converter approach that consumes LSL streams and forwards to OSC. This would be more generic, and would be helpful to people using other systems like muse-lsl on Linux. A simple little CLI tool could be thrown together to handle this. It could be written in a variety of languages, but it would be best to use something that has good support for both LSL and OSC. C# and .NET Core could work well for a cross-platform CLI, and one could utilize the LSL C# bindings https://github.com/labstreaminglayer/liblsl-Csharp and the C# OSC library https://github.com/dastevens/CoreOSC.

I don't have much time to work on something like this right now, but perhaps you could reach out to the NeuroTechX community (https://neurotechx.com/slack/) and find some people who would be interested in helping out. This would be a decent project for a NeuroTechX hacknight.