kowalej/BlueMuse #28: How do I get data from Muse 2 Headband using Core Bluetooth for iOS?

Shinehah-Gnolaum, 2020-05-12:
Would you be willing to help me with information that would help me get data from Muse 2 Headband using Core Bluetooth in iOS?

kowalej, 2020-05-13:
What info are you seeking specifically?

Shinehah-Gnolaum, 2020-05-13:
I thought you could tell me where you found all this information you have on the Muse 2 Headband. 

Muse 2 doesn't seem to use standard Bluetooth. The UUID for the service does not exist in the GATT specification for services. The UUIDs for the characteristics are not standard UUIDs like the UUIDs for characteristics in the GATT specifications for characteristics. 

The main problem I have is that the didUpdateValueFor callback function in Core Bluetooth isn't firing. That's the function that is supposed to catch the data coming from the Muse 2.

kowalej, 2020-05-18:
All of the Bluetooth GAAT UUIDs are listed in the Constants.cs file. https://github.com/kowalej/BlueMuse/blob/master/BlueMuse.App/Misc/Constants.cs.

You have to write specific commands to 0000fe8d-0000-1000-8000-00805f9b34fb to start / stop the streaming within the headband. For example, you have to write these bytes ({ 0x02, 0x64, 0x0a }) to initiate the stream starting, otherwise you will not receive any data on your subscribed channels in your iOS software.

Closing this issue, since it isn't really related to BlueMuse.

closed 2020-05-18 by kowalej

Shinehah-Gnolaum, 2020-05-18:
Ok. Thanks for the information. Could you tell me how you got that information?

kowalej, 2020-05-18:
Use BLE discovery tools and play around. Nordic has some decent tools https://www.nordicsemi.com/Software-and-tools/Development-Tools.