kowalej/BlueMuse #29: can't stream specific muse via command line

chrislos, 2020-05-12:
Hey everyone,
I'm currently trying to autostream two Muse 2016 Devices via command line.

start bluemuse://start?startall
start bluemuse://stop?stopall

works properly.

But as soon as I try to acess a single Muse it doesn't start streaming. 
I tryd the following commands for name and mac adresses. None of these worked for me.

start bluemuse://start?addresses={"Muse-E4DC"}
start bluemuse://start?addresses={Muse-E4DC}
start bluemuse://start?addresses={00:55:da:b0:e4:dc}
start bluemuse://start?addresses={"00:55:da:b0:e4:dc"}
start bluemuse://start?addresses={00-55-da-b0-e4-dc}
start bluemuse://start?addresses={"00-55-da-b0-e4-dc"}

Did I miss anything?
Thanks in advance,
Christian

kowalej, 2020-05-13:
Try dropping the curly braces.

chrislos, 2020-05-14:
boom. worked. Thanks!

closed 2020-05-18 by kowalej