kowalej/BlueMuse #30: Not able to stream from muse 2.

Rakesh420, 2020-05-14:
I am not able to stream from the 2.0.0.0 version as it shows offline always and I installed the 1.1.1.0 version and it crashes as soon as I start to stream.

kowalej, 2020-05-18:
Please give your Muse a try with the official Muse meditation app for iOS or Android, to rule out potential hardware problems.

Rakesh420, 2020-05-19:
I use my muse regularly with the mind monitor app and I've use the official app. The hardware works fine.

kowalej, 2020-05-25:
Can you post your logs? Finding them is described here: https://github.com/kowalej/BlueMuse#logs

rakeshbci, 2020-09-13:
2020-09-13 09:48:30.874 [Information] BlueMuse started.
2020-09-13 09:48:39.782 [Error] Unexpected error get GATT characteristics with GATT device service (UUID=0000fe8d-0000-1000-8000-00805f9b34fb).
System.Exception: Operation aborted (Excep_FromHResult 0x80004004)
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at SharedLibrary!<BaseAddress>+0x4d5c60
   at SharedLibrary!<BaseAddress>+0x4d5be8
   at SharedLibrary!<BaseAddress>+0x4d5ba7
   at BlueMuse!<BaseAddress>+0x50cc2f
2020-09-13 09:48:39.788 [Error] Cannot get device info due to null or empty GATT characteristics.
2020-09-13 09:49:24.516 [Information] BlueMuse started.
2020-09-13 09:49:24.612 [Information] BlueMuse started.
2020-09-13 09:49:40.100 [Error] Exception during toggle stream (start=False). Exception message: ObjectDisposed_Generic.
System.ObjectDisposedException: ObjectDisposed_Generic
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at SharedLibrary!<BaseAddress>+0x4d5c60
   at SharedLibrary!<BaseAddress>+0x4d5be8
   at SharedLibrary!<BaseAddress>+0x4d5ba7
   at BlueMuse!<BaseAddress>+0x507e36
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at SharedLibrary!<BaseAddress>+0x4d5c60
   at SharedLibrary!<BaseAddress>+0x4d5be8
   at SharedLibrary!<BaseAddress>+0x4d5ba7
   at SharedLibrary!<BaseAddress>+0x50929f
   at BlueMuse!<BaseAddress>+0x4fdad1
2020-09-13 09:50:25.154 [Error] Cannot get GATT characteristics due to unexpected communication error with GATT device service (UUID=0000fe8d-0000-1000-8000-00805f9b34fb). Status: Unreachable. Protocol Error .
2020-09-13 09:50:25.154 [Error] Cannot reset device due to null or empty GATT characteristics.
2020-09-13 09:51:23.101 [Information] BlueMuse started.
2020-09-13 09:51:23.203 [Information] BlueMuse started.
2020-09-13 09:51:59.190 [Information] BlueMuse started.
2020-09-13 09:51:59.301 [Information] BlueMuse started.
2020-09-13 09:53:31.527 [Information] BlueMuse started.
2020-09-13 09:54:20.138 [Error] BlueMuse unhandled exception.
System.Exception: Catastrophic failure (Excep_FromHResult 0x8000FFFF)
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at SharedLibrary!<BaseAddress>+0x4d5c60
   at SharedLibrary!<BaseAddress>+0x4d5be8
   at SharedLibrary!<BaseAddress>+0x4d5ba7
   at BlueMuse!<BaseAddress>+0x607753
2020-09-13 09:54:20.154 [Error] BlueMuse unhandled exception.
System.Exception: Catastrophic failure (Excep_FromHResult 0x8000FFFF)
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at SharedLibrary!<BaseAddress>+0x4d5c60
   at SharedLibrary!<BaseAddress>+0x4d5be8
   at SharedLibrary!<BaseAddress>+0x4d5ba7
   at BlueMuse!<BaseAddress>+0x607753
2020-09-13 09:54:20.154 [Error] BlueMuse unhandled exception.
System.Exception: Catastrophic failure (Excep_FromHResult 0x8000FFFF)
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at SharedLibrary!<BaseAddress>+0x4d5c60
   at SharedLibrary!<BaseAddress>+0x4d5be8
   at SharedLibrary!<BaseAddress>+0x4d5ba7
   at BlueMuse!<BaseAddress>+0x607753
2020-09-13 09:54:20.189 [Error] BlueMuse unhandled exception.
System.Exception: Catastrophic failure (Excep_FromHResult 0x8000FFFF)
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at SharedLibrary!<BaseAddress>+0x4d5c60
   at SharedLibrary!<BaseAddress>+0x4d5be8
   at SharedLibrary!<BaseAddress>+0x4d5ba7
   at BlueMuse!<BaseAddress>+0x607753
2020-09-13 09:54:32.107 [Information] BlueMuse started.
2020-09-13 09:56:00.739 [Error] Exception during toggle stream (start=False). Exception message: ObjectDisposed_Generic.
System.ObjectDisposedException: ObjectDisposed_Generic
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at SharedLibrary!<BaseAddress>+0x4d5c60
   at SharedLibrary!<BaseAddress>+0x4d5be8
   at SharedLibrary!<BaseAddress>+0x4d5ba7
   at BlueMuse!<BaseAddress>+0x507e36
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at SharedLibrary!<BaseAddress>+0x4d5c60
   at SharedLibrary!<BaseAddress>+0x4d5be8
   at SharedLibrary!<BaseAddress>+0x4d5ba7
   at SharedLibrary!<BaseAddress>+0x50929f
   at BlueMuse!<BaseAddress>+0x4fdad1

closed 2026-08-15 by kowalej