kowalej/BlueMuse #31: Support for Muse S?

Venryx, 2020-05-18:
Does this software support the latest version of the Muse headset, "Muse S"?

Someone managed to get Muse S data-streaming working here (issue for another Muse LSL repo): https://github.com/alexandrebarachant/muse-lsl/issues/120

Venryx, 2020-05-18:
Nevermind, I've just confirmed it does work: [image]

closed 2020-05-18 by Venryx

Venryx, 2020-05-18:
Oh also, I found this alternative which claims to work with all Muse headsets (haven't tried it myself yet): https://github.com/krigolson/uvicMuse

kowalej, 2020-05-18:
@Venryx  - that's great news, I guess all that's left to do would be adding the device detection so under "Model" it says MuseS in the GUI.

kowalej, 2020-06-06:
@Venryx if possible, could you test out my new prerelease (v2.1) which should detect the Muse S (it won't show up as "Muse2" any more). https://github.com/kowalej/BlueMuse/releases/download/v2.1/BlueMuse_2.1.0.0.zip

Venryx, 2020-06-06:
Unfortunately, version 2.1.0.0 still thinks it is a Muse2:
[image]

Not sure what method is used to distinguish between Muse2 and MuseS, though it seems you could just parse the device-name, checking whether it starts with "MuseS". (perhaps you're already doing this and the issue just lies somewhere else)

kowalej, 2020-06-06:
Hey sorry I think there was an error in that release code, I uploaded another one - same link, don't want to waste your time, but feel free to try again if possible. Been trying to get this fixed, but unfortunately I don't own a Muse S to check myself :)

Venryx, 2020-06-06:
Yep, it works now:
[image]