kowalej/BlueMuse #39: Question about what EEG elements get streamed

paulabrennan, 2021-01-13:
Hi there - great project - I'm just wondering what EEG gets streamed, is it just raw EEG or the separate bands that Muse produces (absolute alpha, etc). Maybe there is a list of elements somewhere?

kowalej, 2021-01-13:
It's raw voltages, there's data for each channel (sensor) separated. Some of the per-processed data that would be in Muse SDK like the power bands are not available. You would have to do the calculations yourself. The channels are shown in the secondary window when you start the stream.

closed 2021-01-13 by kowalej

paulabrennan, 2021-01-13:
Thanks Jason. My idea was to use those pre-processed channels to calibrate the processing settings in Neuromore. I'm mostly interested in the AUX channel anyway.