kowalej/BlueMuse #40: Refresh Command in Command Line Interface

antonello87, 2021-02-18:
Would it be possible to make also the "Refresh" command available from the command line?

I am asking this because I am using the App with multiple Muses and I've seen that if the first Muse is paired and is streaming then if I try to pair automatically a second Muse, it fails to pair until I hit the refresh button. On the contrary if I am not streaming from the first one the second one pairs without any problem. It seems that the LSL bridge (or something else) somehow blocks the pairing.
Is this an expected behaviour?

Thanks a lot,
Antonello

kowalej, 2021-03-11:
It is not available from the command line at this time. I will consider adding it in future released.

kowalej, 2021-12-17:
This is available in 2.3.0.0: https://github.com/kowalej/BlueMuse/releases/tag/v2.3.0.0.

closed 2021-12-17 by kowalej