kowalej/BlueMuse #41: Muse2 EEG stream crash

AlePax, 2021-02-26:
Don't know if this is related to other crashes people are reporting, but in my case I disabled every stream except EEG.

My setup consists of Muse2 connected to OpenVibe3.0 via LSL and from there a TCP writer sends data to an external app.

I noticed the crashes happen more frequently under OpenVibe2.0 than 3.0 and after 5 minutes (20-30 in the case of 3.0), but that migth be a coincidence. Another coincidence maybe, but I noticed that if I select the LSL stream window crashes occur within few seconds as well.

I connected the device both with internal BT and an USB dongle (Asus BT400), but same result.

[BlueMuse-Log-20210226.log](https://github.com/kowalej/BlueMuse/files/6048555/BlueMuse-Log-20210226.log)
[LSLBridge-Log-20210226.log](https://github.com/kowalej/BlueMuse/files/6048556/LSLBridge-Log-20210226.log)

kowalej, 2021-03-10:
Hey sorry to hear you are having issues, unfortunately this looks like a Windows API issue with the Bluetooth, not sure what I can do...

AlePax, 2021-03-11:
Thanks for the feedback. So, I'm kinda screwed unless I elaborate a workaround and bypass Windows altogether...
I just read somewhere that Muse2 can disconnect when low on battery and the LEDs don't always show correct charge, gonna test a little bit more.

kowalej, 2021-03-11:
You can also check Event Viewer in Windows. After launching, look at Windows > Applications Logs and see what comes up during the time that it crashed.

AlePax, 2021-03-11:
Thanks for the tip, will do!

rschmaelzle, 2021-06-30:
Hi all,  I am also using Muse2, and experiencing some stream crashes, albeit they appear somewhat unpredictable. Usually, though, this happens only after several minutes (5-8). The battery is fully charged. Any leads as to what might cause or remedy this would be very helpful (e.g. could it be Bluetooth interference, or is there anything one could do on the receiver-side)? thank you very much and best, ralf

rschmaelzle, 2021-06-30:
to specify this further, when I look into the logs, one potential issue this might have to do is a complaint about "exception during toggle stream (start=false). Exception message: ObjectDisposed_Generic.  ... and then a few refs to SharedLibrary!..../BlueMuse!<...>   . Not sure this helps.

closed 2026-08-15 by kowalej