kowalej/BlueMuse #43: Aux channel for muse2

danielemarinazzo, 2021-05-20:
Hi,
After reading in the readme that the aux is mute with Muse2 I was giving up, but I checked [EEGedu ](https://github.com/kylemath/EEGEdu)  and it worked.
Looks like musejs is able to see the aux. 
https://github.com/urish/muse-js
Any chance the code can be integrated into muselsl?

kowalej, 2021-07-09:
Does it actually work though? I've tried it and not gotten any data on the channel in the past. Would be trivial to re-enable.

danielemarinazzo, 2021-07-10:
We do get data with [Petal](https://petal.tech/). It would be great to have it re-enabled :)

benbao9749, 2021-11-18:
> Does it actually work though? I've tried it and not gotten any data on the channel in the past. Would be trivial to re-enable.

We would really appreciate it if the AUX channel could be an option to include! This issue can be quite important for our research :)

kowalej, 2021-11-18:
I'll try to add it back in the next week.

benbao9749, 2021-11-18:
> I'll try to add it back in the next week.

Thank you so much!

kowalej, 2021-11-22:
@benbao9749 Try this out and let me know if it works: 

https://github.com/kowalej/BlueMuse/releases/download/v2.3.0.0-pre/BlueMuse_2.3.0.0_Debug_Test.zip

benbao9749, 2021-11-23:
> @benbao9749 Try this out and let me know if it works:
> 
> https://github.com/kowalej/BlueMuse/releases/download/v2.3.0.0-pre/BlueMuse_2.3.0.0_Debug_Test.zip

thank you very kindly for working on version 2.3.0.0 to include the aux channel!! 

First, in testing so far, we find that the aux channel data that is streaming to our PC is recognized as NaN -- Not a Number. We will look into this further on our end, but is there something on your side that can also be looked into in this case? With the same muse headset we do detect aux input using the mindmonitor phone application that looks similar to headset electrodes. Data is also streaming normally for the 4 headset electrodes on your version 2.3.0.0 but  only the aux channel is showing as NaN

Second, just for your consideration, would it be helpful for your users to be able to toggle on/off the streaming of aux channel in the settings (as for ppg, gyro, etc)?  As far as our own purposes though, if toggled on, we would appreciate your leaving the aux 5th stream as currently bundled with the 4 others from the headset, rather than being streamed separately, as our software expects the aux data to occur bundled together with the other 4 when it is present. 

Thanks again for your work on BlueMuse -- it is really a great resource for our research!!

kowalej, 2021-11-24:
I set it to NaN if no data came through from Bluetooth for aux, but the other channels are ready to go. Testing with my Muse2 I get no Bluetooth notifications from AUX channel, however I don't have anything plugged into AUX. I wonder if theres a command required to enable the data...

kowalej, 2021-11-24:
@benbao9749 - turns out there is indeed a command needed to enable certain channels. Basically there's different preset "modes". Some include PPG, some EEG + AUX, some just EEG. Anyways, I pushed a test version at same URL as before, I think AUX will work now.

kowalej, 2021-11-24:
I think there's a bug in what I did, since PPG seems to be disabled now, anyways, I think this is the solution, just have to get preset right for the given scenarios.

kowalej, 2021-12-17:
Should work, released now: https://github.com/kowalej/BlueMuse/releases/download/v2.3.0.0/BlueMuse_2.3.0.0.zip

closed 2021-12-17 by kowalej