kowalej/BlueMuse #44: Build Errors with LSLBridge project in VS Community 2019

aung2phyowai, 2021-07-08:
Dear all, I faced the following errors to build the project download from github using VS Community 2019. All 6 errors came from LSLBridge project and 1 warning from BlueMuse.App. As I'm new to VS development, May I know what's the root cause of these problems? How I can  resolve this build issues?

[image]

kowalej, 2021-07-09:
Please check this: https://docs.microsoft.com/en-us/windows/uwp/porting/desktop-to-uwp-enhance.

swatimshah, 2024-09-24:
I faced the same issue. I went through the link shared above. In essence I added the following two packages in "LSLBridge.csproj" and things started building fine (of course after creating a certificate).

<Reference Include="Windows.Foundation.UniversalApiContract">
      <HintPath>C:\Program Files (x86)\Windows Kits\10\References\10.0.19041.0\Windows.Foundation.UniversalApiContract\10.0.0.0\Windows.Foundation.UniversalApiContract.winmd</HintPath>
</Reference>
<Reference Include="Windows.Foundation.FoundationContract">
      <HintPath>C:\Program Files (x86)\Windows Kits\10\References\10.0.19041.0\Windows.Foundation.FoundationContract\4.0.0.0\Windows.Foundation.FoundationContract.winmd</HintPath>
</Reference>

kowalej, 2026-08-15:
Application has been modernized as of 2.5.0.0, and LSL bridge removed. It should be much easier to setup and build now.

closed 2026-08-15 by kowalej