kowalej/BlueMuse #45: Streaming from multiple Muses to same computer

sunshineinsandiego, 2021-07-15:
Hi - Wondering if BlueMuse allows for streaming multiple muses simultaneously to the same computer and then recording all streams to LabRecorder. Does this require opening multiple instances of BlueMuse i.e. one instance for each headset, or is this possible using a single instance of BlueMuse to multiple headsets? Understand that in the FAQs there may be issues with time synchronicity for streaming multiple headsets. Advice? Thanks!

mmestucci, 2021-07-16:
Hi @sunshineinsandiego yes, BlueMuse allows to connect and to stream multiple devices. 

I'm trying a setup with 2x Muse 2 connected to the same pc using BlueMuse and it works. 

Just pay attention about the rate: I tested it with different machines and it works well just on powerful Dell laptop ...
[image]

sunshineinsandiego, 2021-07-16:
Got it, thank you! Appreciate the help.

closed 2021-07-16 by sunshineinsandiego