kowalej/BlueMuse #50: Connectivity issue with Muse-2 and BlueMuse 2.2.0.0

akchoj, 2021-11-24:
Muse model MU-02.  Using a laptop with Sabrent BT-UB40 dongle as recommended.  Turned laptop BT off, only dongle BT was on.  No connection at all.

Log file states:
2021-11-23 21:20:02.036 [Information] BlueMuse started.
2021-11-23 21:30:02.047 [Information] BlueMuse started.

With dongle BT off and Laptop BT on, BlueMuse detects Muse, but Muse stays offline-connection error.  Again log indicates:
2021-11-23 21:33:57.257 [Information] BlueMuse started.
No error reported.

With Muse off, laptop BT on, starting BlueMuse, it puts Muse in list of devices, but offline.

kowalej, 2021-11-24:
Check the BT settings in Windows and make sure the Muse is not paired. You can also try the "Always Pair" setting in BlueMuse to see if that helps.

kowalej, 2021-11-24:
I'm always skeptical of these dongles though and how they interact with Windows APIs - I couldn't get the one I had working, but my motherboard built in works.

akchoj, 2021-11-24:
Thanks for the reply.  Okay, here is what I tried, and here is what actually worked.  Tried two Asus laptops with onboard BT.  Neither one worked.  Tried a Dell desktop with a Qualcomm QCA9377 onboard blue tooth.  That worked.  That Dell had Windows 10 Home Build 19042.  Asus laptops had builds 19043 and 19044.  Rolled back update to 19042 on Asus laptop.  That didn't work.  The one Asus laptop I had access to has a RealTek bluetooth onboard, and its drivers were current.  Went out bought an Asus ASB BT-500 5.0 dongle.  Disabled onboard RealTek bluetooth adapter, and installed the Asus dongle.  This also didn't work (neither on the Dell nor the Asus laptop).  Funny enough the BT-500 USB 5.0 dongle also uses a RealTek chipset.  Exchanged Asus dongle for a Startech Mini USB BT 4.0 (USBBT1EDR4).  This dongle runs of a Qualcomm (CSR- CSR8510A06) BT chipset.  With the onboard BT disabled in Device Manager,  this dongle worked once Windows installed the drivers.  Didn't need to use "Always Pair" setting.  My guess is an incompatibility issue with Realtek BT chipsets (not sure what chipsets they are as I haven't looked into them) and Muse 2.  Willing to bet other Asus laptop also runs a RealTek BT adapter onboard.  It's important to note that you need to disable the non-compatible BT adapter that is on the Laptop motherboard, before the dongle will work (that's my experience anyways).

kowalej, 2021-12-17:
Wow pretty interesting. Not sure what I could do on my end though, since I'm using the official Windows BLE APIs. This sounds like either a driver issue with Realtek or some obscure Bluetooth incompatibility with the Muse's Bluetooth chip.

Thanks for your input.

closed 2021-12-17 by kowalej