kowalej/BlueMuse #52: Cannot install on windows 10.

Zibri, 2022-03-21:
```
PS C:\Users\Zibri\Downloads\BlueMuse_2.3.0.0> .\Install.ps1                                                             Found bundle: C:\Users\Zibri\Downloads\BlueMuse_2.3.0.0\BlueMuse_2.3.0.0_x86_x64.appxbundle
Found certificate: C:\Users\Zibri\Downloads\BlueMuse_2.3.0.0\BlueMuse_2.3.0.0_x86_x64.cer

Before installing this app, you need to do the following:
        - Install the signing certificate
The certificate was successfully installed.

Installing app...
Found dependency package(s):
C:\Users\Zibri\Downloads\BlueMuse_2.3.0.0\Dependencies\x86\Microsoft.NET.Native.Framework.2.2.appx
C:\Users\Zibri\Downloads\BlueMuse_2.3.0.0\Dependencies\x86\Microsoft.NET.Native.Runtime.2.2.appx
C:\Users\Zibri\Downloads\BlueMuse_2.3.0.0\Dependencies\x86\Microsoft.VCLibs.x86.14.00.appx
C:\Users\Zibri\Downloads\BlueMuse_2.3.0.0\Dependencies\x86\Microsoft.VCLibs.x86.14.00.Desktop.appx
C:\Users\Zibri\Downloads\BlueMuse_2.3.0.0\Dependencies\x64\Microsoft.NET.Native.Framework.2.2.appx
C:\Users\Zibri\Downloads\BlueMuse_2.3.0.0\Dependencies\x64\Microsoft.NET.Native.Runtime.2.2.appx
C:\Users\Zibri\Downloads\BlueMuse_2.3.0.0\Dependencies\x64\Microsoft.VCLibs.x64.14.00.appx
C:\Users\Zibri\Downloads\BlueMuse_2.3.0.0\Dependencies\x64\Microsoft.VCLibs.x64.14.00.Desktop.appx
Add-AppxPackage : Deployment failed with HRESULT: 0x80073CF3, Package failed updates, dependency or conflict
validation.

Windows cannot install package 07220b98-ffa5-4000-9f7c-e168a00899a6_2.3.0.0_neutral_~_965tqh4daexne because a
different package 07220b98-ffa5-4000-9f7c-e168a00899a6_1.1.1.0_neutral_~_te84kyc8jyx80 with the same name is already
installed. Remove package 07220b98-ffa5-4000-9f7c-e168a00899a6_1.1.1.0_neutral_~_te84kyc8jyx80 before installing.

NOTE: For additional information, look for [ActivityId] 51048203-3d3d-0002-809b-04513d3dd801 in the Event Log or use
the command line Get-AppPackageLog -ActivityID 51048203-3d3d-0002-809b-04513d3dd801

At C:\Users\Zibri\Downloads\BlueMuse_2.3.0.0\Add-AppDevPackage.ps1:487 char:13
+             Add-AppxPackage -Path $DeveloperPackagePath.FullName -Dep ...
+             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : WriteError: (C:\Users\Zibri\..._x64.appxbundle:String) [Add-AppxPackage], IOException
    + FullyQualifiedErrorId : DeploymentError,Microsoft.Windows.Appx.PackageManager.Commands.AddAppxPackageCommand

Error: Could not install the app.  To ensure security, please consider uninstalling the signing certificate until you can install the app.  Instructions for doing so can be found here:
http://go.microsoft.com/fwlink/?LinkId=243053
Press Enter to continue...:
```

kowalej, 2022-03-21:
Make sure you are running `InstallBlueMuse.ps1`, since it handles removal of old package first. That is the problem you are having.

If it doesn't work, just manually uninstall current BlueMuse version.

Zibri, 2022-03-21:
I just did:

```
PS C:\Users\Zibri\Downloads\BlueMuse_2.3.0.0> .\InstallBlueMuse.ps1
Removed BlueMuse 1.1.1.0
Running: C:\Users\Zibri\Downloads\BlueMuse_2.3.0.0\.\Install.ps1
Found bundle: C:\Users\Zibri\Downloads\BlueMuse_2.3.0.0\BlueMuse_2.3.0.0_x86_x64.appxbundle

Installing app...
Found dependency package(s):
C:\Users\Zibri\Downloads\BlueMuse_2.3.0.0\Dependencies\x86\Microsoft.NET.Native.Framework.2.2.appx
C:\Users\Zibri\Downloads\BlueMuse_2.3.0.0\Dependencies\x86\Microsoft.NET.Native.Runtime.2.2.appx
C:\Users\Zibri\Downloads\BlueMuse_2.3.0.0\Dependencies\x86\Microsoft.VCLibs.x86.14.00.appx
C:\Users\Zibri\Downloads\BlueMuse_2.3.0.0\Dependencies\x86\Microsoft.VCLibs.x86.14.00.Desktop.appx
C:\Users\Zibri\Downloads\BlueMuse_2.3.0.0\Dependencies\x64\Microsoft.NET.Native.Framework.2.2.appx
C:\Users\Zibri\Downloads\BlueMuse_2.3.0.0\Dependencies\x64\Microsoft.NET.Native.Runtime.2.2.appx
C:\Users\Zibri\Downloads\BlueMuse_2.3.0.0\Dependencies\x64\Microsoft.VCLibs.x64.14.00.appx
C:\Users\Zibri\Downloads\BlueMuse_2.3.0.0\Dependencies\x64\Microsoft.VCLibs.x64.14.00.Desktop.appx

Success: Your app was successfully installed.
Press Enter to continue...:
PS C:\Users\Zibri\Downloads\BlueMuse_2.3.0.0>
```

Zibri, 2022-03-21:
The app start. Pairs.
As I do START STREAMING it closes.

kowalej, 2022-03-21:
Try using always pair option in settings.

kowalej, 2022-03-21:
I'll close this issue since your install works now. Please open another issue for other problems.

closed 2022-03-21 by kowalej