kowalej/BlueMuse #53: Application exits (closes) as soon as I press start streaming

Zibri, 2022-03-21:
I also enabled the experimental always pair in the settings.
It pairs and gets battery status and device name.
[image]
[image]

Zibri, 2022-03-21:
2022-03-21 19:34:15.466 [Information] BlueMuse started.
2022-03-21 19:36:27.194 [Error] BlueMuse unhandled exception.
System.UnauthorizedAccessException: Access is denied. (Exception from HRESULT: 0x80070005)
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw() + 0x21
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task) + 0x70
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task) + 0x38
   at System.Runtime.CompilerServices.TaskAwaiter.ValidateEnd(Task) + 0x17
   at System.Runtime.CompilerServices.TaskAwaiter.GetResult() + 0xb
   at BlueMuse.Bluetooth.BluetoothManager.<ActivateLSLBridge>d__28.MoveNext() + 0xf9
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw() + 0x21
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task) + 0x70
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task) + 0x38
   at System.Runtime.CompilerServices.TaskAwaiter.ValidateEnd(Task) + 0x17
   at System.Runtime.CompilerServices.TaskAwaiter.GetResult() + 0xb
   at BlueMuse.Bluetooth.BluetoothManager.<StartStreaming>d__30.MoveNext() + 0x12e
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw() + 0x21
   at System.Runtime.CompilerServices.AsyncMethodBuilderCore.<>c.<ThrowAsync>b__7_0(Object) + 0x1e
   at System.Action`1.Invoke(T) + 0x28
   at System.Threading.WinRTSynchronizationContext.Invoker.InvokeCore() + 0x33

kowalej, 2022-03-21:
`Access is denied.` seen this before from other users, not sure how to help. If you are using Bluetooth dongle, I'd try a different one, other than that I don't have much to offer.

Zibri, 2022-03-21:
It has nothing to do with bluetooth. The connection with the device works.
It's your program not able to create the streaming.
As a matter of fact this test application works perfectly: https://muse-eeg-app.firebaseapp.com/

closed 2026-08-15 by kowalej