kowalej/BlueMuse #59: Installation issue

yu-zhengzheng, 2022-11-29:
When executing file in powershell on win 11 with the command PS ...\BlueMuseMaster\Tools> .\InstallBlueMuse.ps1
It gives the following error:

& : The term '...\BlueMuseMaster\tools\.\Install.ps1' is not recognized as the name of a cmdlet,
function, script file, or operable program. Check the spelling of the name, or if a path was included, verify that
the path is correct and try again.
At ...\BlueMuseMaster\tools\InstallBlueMuse.ps1:39 char:3
+ & $InstallScript
+   ~~~~~~~~~~~~~~
    + CategoryInfo          : ObjectNotFound: (E:\PyCharm\Proj...s\.\Install.ps1:String) [], CommandNotFoundExceptio
   n
    + FullyQualifiedErrorId : CommandNotFoundException

I tried to manually install but executing the script only opens a terminal window that quickly disappears.

NathanielRose, 2023-02-09:
Having the same thing, I believe this release is incompatible with windows 11

kowalej, 2023-02-09:
I'll try making a Windows 11 build. I'm on Window 10 still, but it should be easy to re-target and cut a new release.

kowalej, 2023-02-09:
@NathanielRose can you try this and let me know. https://github.com/kowalej/BlueMuse/releases/download/v2%2C4.0.0/BlueMuse_2.4.0.0.zip

kowalej, 2023-02-10:
@yu-zhengzheng - feel free to try that build as well:  https://github.com/kowalej/BlueMuse/releases/download/v2%2C4.0.0/BlueMuse_2.4.0.0.zip

NathanielRose, 2023-03-01:
@kowalej This worked on my machine 👍🏿

closed 2023-03-01 by kowalej