kowalej/BlueMuse #63: EEG sampling frequency issue in v2.4.0.0

anjula04, 2023-11-09:
Hi,

I experience a large variation in the sampling frequency after few minutes and when I move the muse. It varies from ~700 Hz to ~10 Hz. But after it drops down it never recovers. This is similar to https://github.com/kowalej/BlueMuse/issues/54#issue-1195122622.

Any solutions for this?
Will connecting through the BLED112 sort this issue?

Thanks.

kowalej, 2023-11-12:
Aside from connectivity issues I've seen this happen on systems with high resource usage, maybe check your free memory/ CPU. Other than that there's not much I could advise. I think it may be worth trying the dongle though as you said, as it could be your BLE hardware that's causing problems.

anjula04, 2023-11-20:
Hi, 
I tried with the BLED112 and sampling rates were stable near 256 Hz! Thank you for the suggestion.

However, I was able to stream data only through [uvicMUSE](https://github.com/krigolson/uvicMuseBLED). BlueMuse did not detect the device. What should I do to get this up and running?

kowalej, 2023-11-27:
@anjula04 - I'll have to close this since I don't have a solution for you using BlueMuse. The app is built to use the native Windows BLE stack and not integrate with dongles directly.

anjula04, 2023-11-28:
@kowalej I was able to get consistent connection through the native Windows BLE when I used https://petal.tech/metrics-api. It give a consistent sampling rate of 256 Hz. The connection drops once in a while but automatically reconnects within few seconds.