kowalej/BlueMuse #64: streaming 4 muse2 devices from bluemuse in a single laptop

aung2phyowai, 2023-11-12:
This is not an issue. Like to seek your advice on whether 4 muse2 devices can be connected and stream EEG+PPG data using single bluemuse application running at a single laptop. What are additional requirements and setup to enable multiple devices streaming from a single laptop?

kowalej, 2023-11-12:
The app supports multiple devices already. You may be bottlenecked by the hardware, but on a decent specced laptop with a good Bluetooth chip, I don't see you having problems with 4 devices. I've streamed from several devices at once before with no issues, mind you this was just EEG channels. However the other channels in addition shouldn't have too much impact since they are at a lower data rate.

kowalej, 2023-11-27:
Closing this as it's not an issue.

closed 2023-11-27 by kowalej