kowalej/BlueMuse #65: BlueMuse streaming and muselsl streaming give different results

swatimshah, 2023-11-27:
Hello, I am recording the Muse-2 signals in a CSV file. I realized that for electrodes TP9 and AF8, the amplitude is recorded as '0' after certain intervals. Please browse through the below recorded csv file.
[https://drive.google.com/file/d/1INOLh4zTbTAVMkOPXfj0Ma_hSZJLbj-y/view?usp=sharing](https://drive.google.com/file/d/1INOLh4zTbTAVMkOPXfj0Ma_hSZJLbj-y/view?usp=sharing)

I am using the below commands for streaming and recording the above signals.
**muselsl stream --name Muse-94XX
muselsl record --duration 230**

This phenomenon (the signal amplitude becoming zero, occasionally) does not occur when I use BlueMuse for streaming.
Could you please check and confirm the difference between "BlueMuse streaming" and "muselsl streaming"? 
Is BlueMuse streaming the only right way to go?

kowalej, 2023-11-27:
`BlueMuse` and `muselsl` are separate applications that can be used together in a way, let me explain:

`muselsl` is written in Python and by default uses a Python library to stream the Bluetooth data from your Muse. It can also be used to plot and visualize EEG data, among other things.

`BlueMuse` is a separate Windows GUI program which uses the native Windows Bluetooth stack for streaming. It also has its own code for LSL streaming, although it should be using more or less the same underlying library as `muselsl` for this (both binding to a C  lib).

Therefore there may be differences in streaming performance between these applications, which may correspond to the zero signals in your recording. The solution for you may be to use `BlueMuse` as the "backend" for `muselsl` streaming. You can do this by running your stream command like this:
`muselsl stream --name Muse-94XX --backend bluemuse`

This will tell `muselsl` not to use it's native streaming, and instead pass the streaming commands off to `BlueMuse`. It would then record using the lsl stream from there, so you can still run `muselsl record --duration 230` as normal.

swatimshah, 2023-11-28:
Hello, Thank you very much for your detail response. It helps a lot in understanding the way BlueMuse and muselsl work. However, I still have a couple of more questions. 

My requirement is to use the EEG stream from muselsl/BlueMuse to model the brain signals for BCI purpose. My goal is to get a handle over EEG stream to model the signals. 

In this process, I am analyzing the "muselsl" code. I realized that in "native streaming" the value of "LSL_EEG_CHUNK" is zero few times. That is why the zero signals are recorded for all the 12 samples of the chunk. If I skip this "zero valued chunk", will that be an accurate EEG signal stream for modelling?  

However, if I should go "BlueMuse way", is there a way to get a handle over EEG stream (apart from modifying the c-sharp code and build it again for my requirement)?

kowalej, 2023-11-28:
BlueMuse still uses LSL to stream the data - so if you're asking could you write your own code to handle the stream, the answer would be yes.

If you are using the BlueMuse backend with muselsl and still seeing these zero chunks, it sounds like there's packet loss from the headset, or possibly it's not worn securely to ensure solid electrode contact (although that probably wouldn't result in zero values).

Since this is not a specific issue with BlueMuse I'm going to close it.

closed 2023-11-28 by kowalej

swatimshah, 2023-11-29:
I thought I will just mention this as an FYI here. I changed python version from 3.8 to 3.9 and muselsl version from 2.2.0 to 2.2.2. The issue of "zero chunks" in muselsl "native stream" is solved for rest of the electrodes except TP9. 

I tested this twice, with 1 min data. 

It is not a BlueMuse issue. I will raise this issue on muselsl forum.  Thanks a lot for all your help.