kowalej/BlueMuse #68: Error - while building LSLBridge

swatimshah, 2024-09-20:
I am trying to build the BlueMuse project in Visual Studio. I am getting below error message 
**The type forwarder for type 'Windows.ApplicationModel.AppService.AppServiceConnection' in assembly 'Windows' causes a cycle.**

This error arises while building LSLBridge project. 

I am performing the following steps:
- Clean nuget cache
- Clean obj and bin folders
- Clean solution
- Build solution

Could you please let me know where I am possibly going wrong? 
Thanks in advance!

swatimshah, 2024-09-24:
I could figure out a solution for successful build after going through the issue 44. In essence I added the following two packages in "LSLBridge.csproj" and things started building fine (of course after creating a certificate).

`
<Reference Include="Windows.Foundation.UniversalApiContract">
      <HintPath>C:\Program Files (x86)\Windows Kits\10\References\10.0.19041.0\Windows.Foundation.UniversalApiContract\10.0.0.0\Windows.Foundation.UniversalApiContract.winmd</HintPath>
</Reference>
`
`
<Reference Include="Windows.Foundation.FoundationContract">
      <HintPath>C:\Program Files (x86)\Windows Kits\10\References\10.0.19041.0\Windows.Foundation.FoundationContract\4.0.0.0\Windows.Foundation.FoundationContract.winmd</HintPath>
</Reference>
`

kowalej, 2026-08-15:
Application has been modernized as of 2.5.0.0, and LSL bridge removed. It should be much easier to setup and build now.

closed 2026-08-15 by kowalej