kowalej/BlueMuse #69: cannot install BlueMuse2.4 Windows 11

amisepa, 2024-10-11:
1. Tried the InstallBlueMuse.ps1
Opens up this error in my text editor:
```
function PrintMessageAndExit($ErrorMessage, $ReturnCode)
{
    Write-Host $ErrorMessage
    if (!$Force)
    {
        Pause
    }
    exit $ReturnCode
}

$ErrorCodes = Data {
    ConvertFrom-StringData @'
    Success = 0
    NoScriptPath = 1
'@
}

$ScriptPath = $null
try
{
    $ScriptPath = (Get-Variable MyInvocation).Value.MyCommand.Path
    $ScriptDir = Split-Path -Parent $ScriptPath
}
catch {}

if (!$ScriptPath)
{
    PrintMessageAndExit "Can't get current directory" $ErrorCodes.NoScriptPath
}

$Package = Get-AppxPackage -name '07220b98-ffa5-4000-9f7c-e168a00899a6'
if($Package){
	Remove-AppxPackage $Package.PackageFullName;
	Write-Host ("Removed BlueMuse " + $Package.Version)
}

$InstallScript = Join-Path $ScriptDir "./Install.ps1"
Write-Host ("Running: " + $InstallScript)
& $InstallScript

2. then I tried manual install and get this error when trying to install Microsoft.NET.Native.Framework.2.2 (after successfully installing certificates)

Any help greatly appreciated.

```

SagHack, 2024-12-02:
1. That is not an error, that is the content of the script. Just run in the powershell as is in the installation document:  .\InstallBlueMuse.ps1

closed 2024-12-15 by kowalej

amisepa, 2024-12-18:
Sorry for the slow response. I'm pretty sure I run it in the PowerShell. I'll try again and let you know.