kowalej/BlueMuse #72: Difficulty Installing Blue-Muse

ishanb200, 2025-04-06:
I've tried both the manual and automatic installations, but they both give me different errors n problems :(

For the Manual installation, when trying to install the last 2 files under dependencies (Microsoft.NET.Native.Framework2.2 and Microsoft.NET.Native.Runtime2.2 both x64) it tells me to stop running "Microsoft.GamingServices" 
However, I find no such file when under task manager. If it doesn't exist, how tf am I expected to end the task..

The automatic installation is even worse because when I run "InstallBlueMuse.ps1" in powershell, it does something funny. Immediately opens powershell and closes. Now I don't even get any popups to try n troubleshoot so what am I even supposed to do here...

[image]

closed 2025-04-08 by ishanb200

kowalej, 2025-04-09:
Did you get it working @ishanb200 ?

ishanb200, 2025-04-10:
> Did you get it working [@ishanb200](https://github.com/ishanb200) ?

turns out I needed to close Microsoft.GamingServices